intergalaktik/ulx4m #8: multiboot and programming issues

emard, 2021-12-09:
I compiled multiboot DFU loader and if holding BTN1 and
plugging USB, it enumerates as DFU device

It accepts flashing some bitstream but then user bitstream doesn't work
assign led = 4'b0101; // and all LEDs off

multiboot bitstream itself uses those LEDs
but when control is passed to user bitstream,
then LEDs doesn't work

Content of user bitstream area is ok I uploaded as
dfu-utill -a0 -D bitstream.bit
and then downloaded as
dfu-util -a0 -D readback.raw

comparing bitstream.bit and readback.raw, beggingings
are the same content, only readback is longer than bitstream.bit
as it runs until end of flash

after this I can't even upload anything temporary with fujprog, this happens:
fujprog build-tmp/passthru.bit
ULX2S / ULX3S JTAG programmer v4.8 (git cc3ea93 built May 23 2021 17:03:05)
Copyright (C) Marko Zec, EMARD, gojimmypi, kost and contributors
Using USB cable: ULX3S FPGA 85K v3.0.8
Programming: 96% | Received and expected data do not match!
TDO: 00000000 Expected: 00000100 mask: 00002100
Line 5706: Operation not permitted

Failed.

emard, 2021-12-09:
model=ulx4m DEVICE=um-85k:
this must be used for bootloader to jump to user bitstream
without this, even normal programming will fail if bootloader is loaded.
this can also be used to apply non-serdes bootloader bitstream to serdes 85F chip
--input-idcode  0x01113043 \
--output-idcode 0x01113043 \

$(BUILD_TMP)/multiboot.img: $(BUILD_TMP)/$(PROJ).bit build-tmp/passthru.bit
        ecpmulti                           --input $(BUILD_TMP)/$(PROJ).bit \
 --input-idcode  0x01113043 \
 --output-idcode 0x01113043 \
          --address $(USER_BITSTREAM_ADDR) --input build-tmp/passthru.bit \
          --flashsize 128 \
          --output $@

goran-mahovlic, 2022-01-04:
I will close this one - as it works with quick fix. And it is reported to YosysHQ

closed 2022-01-04 by goran-mahovlic