intergalaktik/ulx4m #49: extra io - compatibility with radxa cm3

cz172638, 2023-05-26:
radxa's cm3 is compatible with rpi's cm4, but extra connector gives more io connectivity. having ulx4m compatible with radxa's cm3 module, we can leverage io boards for cm3.
[comparison between cm4 (rpi) and cm3 (radxa)](https://wiki.radxa.com/Rock3/CM3/vsCM4)

goran-mahovlic, 2023-05-26:
Tnx for the idea!

Board is compatible but as you noticed one connector is missing to have more pins. 

Problem is that I do not have that many available pins (I have only few) and if I would use those I would need to switch to more then 6 layers board that would be more expensive.
Also in datasheet of those hirose connector it is stated that it is adequate only for one connector - CM4 already have 2 - so the third one will make things even harder for production/users.
We would also block usage on many IO boards as most have components bellow module.
I will think about it more, but from my perspective it would not worth the effort as we would lose more then we would get...

And in future for some other chip that would have more IO-s that may be the idea to try out...

goran-mahovlic, 2025-03-12:
Closing as not possible

closed 2025-03-12 by goran-mahovlic